tompi/cheapino #82: Entire top row not working

cyberlord-coder-228, 2024-10-01:
Hi, I'm in the process of building my first cheapino, and I've noticed that entire top row and outer most thumb key on the left side of the board don't register.

I haven't touched right side yet. Only soldered the diodes and MCU in; and flushed the MCU. (Haven't even clipped diodes' legs off, as you can see in the picture))

Shortening keys in the bottom & middle rows and 2 of the 3 thumb keys  give correct key presses, but top row does not register. I've tried shortening some pin pairs on the MCU directly, and I managed to get q, w, and e.

In the troubleshooting section it is mentioned that entire row not working is probably due to bad connections on the MCU or rj-45. Since I only use one part of the board, I dont think it's rj-45 related. Didn't notice any shorts in it's place either.
I've rechecked and redone all the MCU pins (and socket pins) several times, but the issue persisits. Granted, the soldering job is very sloppy, as it is my first soldering project, but I couldn't detect any shorts or any other obvious issues with it.

All diodes are also directed "line to the square".

I didn't manage to get the veil website working, and just tested the keyboard by entering some text, if that's relevant.

So, what direction should I take to troubleshoot it further?

// Those are the best quality pics that my phone can make //
[image]
[image]
[image]
[image]
[image]

Btw, thanks for awesome project :3

tompi, 2024-10-01:
The Lone thumb key not working is probably a defective diode. Check with a multimeter?(or just replace it)

For the row: i suggest you test with the vial fw and yes the matrix tester there. 

Can you trigger the row by shorting 2 and 2 of the bridge pads(just 2 of the lowest pads, dont bridge them)

cyberlord-coder-228, 2024-10-01:
Thank you for such quick reply :3
And thank you for all the advice

As for thumb diode, I don't have tools right now, so I'll check more thoroughly tommorow. But I noticed that shortening one of the closer circles + cirle part of the diode gives some output. Not on this key, though.
[image]

I'm not sure what did you mean about vial. Should I install a native app and fiddle with that a bit?
There's matrix tester's result
[image]

And I'm completely not sure what did you mean about shorting 2 and 2 of the pads.
I tried poking at pins and connections until I got some output) Two big square points, where the hotswaps are to be soldered to, give output. So do diagonal circles (that's how I tested them),  And so do two bottom circles, or two top circles.
[image]

If you meant parts that must be bridged on the right part of the board, then no, no combination that I've tried seem to give any kind of output. I did, however, got a few keys by shortening the rj-45 pins.

I played a bit with shortening pins on the MCU and (apart from pretty sparkes on pins... whait, GND and 5U?... damn) , I think I triggered reset, cause all of the keys stopped working.
I reflushed it, and 1) I noticed that my system did not automatically detected the MCU as a flash drive, as it did the first time when I connected it (the first time I dismissed it as some type of Linux magic, but now I think that something was shortened somewhere, and I fixed it when I resoldered MCU pins; which is good, probably).
And 2) some more buttons don't work now.
[image]

It's... bizzare. But I think reinforces that something is terribly wrong with the MCU

tompi, 2024-10-02:
Okay, sparks are not a good sign :(

You should only short 2 and 2 data pins on the mcu, never ground or power...

But is sounds like the mcu is damaged...
Good thing you socketed the mcu.

I would take out the mcu and try shorting 2 and 2 of rows: gp27-gp29 and gp8, and columns:  gp14,gp15,gp26

If you are not able to completely fill the vial test matrix doing this your mcu is not working...

cyberlord-coder-228, 2024-10-02:
Ohhhh, thats what you meant by 2 and 2...

I did not take out the MCU (I'm intimidated by desoldering it, I barely figured how to solder that beast in), just tried that test as is, by poking (diode legs') pin pairs {27-29} and {14-26}. Everything works :|
[image]

But on the actual keys' pins everything as it was after evertyhing got worse)
// like that
[image]

Another interesting observation is that i cannot trigger the pins by touching solder "volcanoes", only diode legs. Not sure if thats intended behavior or not.

Oh and not sure if I overstated the sparks, that was _just_ arcs, not New Year type performance

tompi, 2024-10-02:
Ok. Thats a good sign that you can trigger them. The mcu should be fine then. 

You probably have some bad soldering, or maybe the legs that go into the socket Are bent/not long enough? You did socket the mcu it seems, so it should pull right out.

Anyway, can you trigger the keys from the underside of the board as well? If not its the soldering of the socket or the socket pins thats the problem. 

Try reflowing on both sides. Use plenty of flux and correct temp, and do not spend too much time...

Another possibility is that you burnt off or scratched off one column trace and one row trace...

tompi, 2024-10-13:
Hey, just made a tool for debugging these cases, check it out if you havent given up:

https://tompi.github.io/cheapino/doc/troubleshooting/routing.html

Should be easy to narrow down which joints to check.

closed 2024-11-10 by tompi