tompi/cheapino #85: Two Rows On My Cheapino V2 Just Dont Work

skie3e, 2024-10-06:
the right half works perfectly but the two middle left rows on my left side dont work at all!

i've re-heated all hot-swap switches and diodes multiple times

would it you know which diode i could switch out to try and fix this?

[image]
(i didnt plug in the right half but it works)

im using through hole diodes

i took the best photos i could, sorry they're so bad ;-;

[image]
[image]
[image]
[image]
[image]

ignore that some of my hot swap switches are melted, soldering is hard :P 

thanks in advance! i love this project!!!!

tompi, 2024-10-06:
Hey, Im working on a visual debugging tool for this, ill give you a sneek peak :)

You should reflow the pin that I drew an arrow at, both columns go into that pin, so that has to be the culprit.
Also, try to use correct temp, flux, and time when soldering.

[image]

skie3e, 2024-10-06:
wow did something weird happen, when i redid that i lost 4 keys on the right hand,, top row, outside moving in
also, now the left-most column triggers the middle column and the thumb cluster triggers the middle left column
not sure but i cant see any brigded connections, im so confused :P 

i currently ave no keyboard because my wire broke so im using on-screen keyboard 😭

tompi, 2024-10-07:
Im afraid this very likely comes down to the soldering.
Take a look at this illustration, and maybe check out some youtube guides:
[image]

The image is from this guide:
https://www.circuitspecialists.com/blog/how-to-soldering-correctly-soldering-101/

tompi, 2024-10-13:
Hey, just made a tool for this, check it out to debug:
https://tompi.github.io/cheapino/doc/troubleshooting/routing.html

closed 2024-11-10 by tompi