tompi/cheapino #87: Single key also fires adjacent key

StarNamer, 2024-10-07:
I thought everything was OK and assembled the case then found that the M key also fires the N key. The N key on its own only fires itself. All other keys are fine. I'm assuming the means there's a short somewhere; does this information indicate where it's likely to be?

Fortunately, I made the switches swappable, but it means taking the case apart...

To add to your gallery...
[image]
As it's my first keyboard, I took the keycaps off an old PU-87 I bought as non-functional on eBay. I chose to print the case in orange to pick up the keycap legend colour,

StarNamer, 2024-10-07:
Fortunately trivial. A poorly trimmed diode leg was pushed against the switch mount when the bottom of the case was put on!

closed 2024-10-07 by StarNamer

tompi, 2024-10-07:
The best kind of issue is the one solved when I discover it ;)

Thanks for the image, looks really good!

P.S. Switching the color of the diode to yellow will probably be cool! (look into qmk keys controlling led, you can control color, brightness and intensity and enable some effects if you like also)