tompi/cheapino #89: anyway to make connection with computer wireless?

jinchoung, 2024-10-10:
hi thomas!

hope all is well with you and yours. still super in-love with the cheapino and it's my daily driver and i can't imagine having to do any amount of typing without it now!

but especially because i'm so dependent on it, i"m wondering more and more if there was a simple way to make the cheapino connect to the pc wirelessly. i don't mind or care about the rj45 connection between left and right sides. that's totally cool. but i do wish there was a way to make the whole keyboard wirelessly connected to the computer.

any ideas? i've looked around to see if they had 2.4ghz dongles but they're all just receivers, not transmitters. same thing with blutooth dongles. of course it would need to be powered too but i wouldn't mind having to solder on a battery.

thanks much!

jin

tompi, 2024-10-10:
Hey. Im working on a wireless solution, but got a bit sidetracked...

In short, the xiao seed ble is pin compatible, so just need to swap mcu with that(mount the ble mcu at the back) and fit a battery plus switch somewhere. 

There really isnt much place for a battery if you keep the rj45 ports?

I was thinking i would drop the encoder on the right for another mcu and use rj45 space for batteries...

jinchoung, 2024-10-10:
oh wow! that sounds like a simple solution! well keep us updated if you get back to that project!

is it really as simple as swapping out the rp2040 then flashing it with the same firmware? you wouldn't have to modify it for bluetooth connection? if so, i can just put in an order for the boards now....

i'd love to be able to update my keyboard to that. personally, i don't mind just have a battery floppy around the kb. at worst, i can just doublesided tape it to the top of one of the rj45 jacks!

thanks again.

jin

tompi, 2024-10-10:
No, it needs to run zmk, and i need to do a minor mod to the scan Code, thats where i left it last time...

jinchoung, 2024-10-10:
ah i see. ok, will stay tuned for that.

but to expand the conversation, i just found (man, you really gotta feed google the right search terms!) that there are kits like this too!

https://www.youtube.com/watch?v=MEMyae7b74I

may actually be a simpler solution although it will probably cost a little more.

jin

azzamsa, 2024-10-12:
> I was thinking i would drop the encoder on the right for another mcu and use rj45 space for batteries...

Is there any reason not to make the case thicker without dropping the encoder?
If you use a tenting case like [this](https://www.printables.com/model/879047-two-case-designs-for-cheapinov2-split-ergo-keyboar), it will fit any battery.

[image]

azzamsa, 2024-11-05:
@tompi Do you have a working wireless version of the Cheapino?

My PCBs have arrived, and I’d like to build them wirelessly. Could you go over the steps and any additional materials I’ll need, beyond what’s listed here: https://github.com/tompi/cheapino/blob/master/doc/orderingguide.md#the-components?

tompi, 2024-11-05:
Yes! I attached the batteries and typed for a short session yesterday at home.

These are the drawbacks:
No encoders(need to hack my own scanning code in zmk for that)
No deep sleep(hence you should have on/off switches to avoid draining batteries)
Tested for a total of 20 minutes on one computer...

You will need 2 seed studio BLE controllers(around 10usd per piece...), 2 batteries and 2 toggle switches.
 
Im gonna use this for work the next 3 days, might find some more snags by then...

Code I used is here:
https://github.com/tompi/cheapinov2-zmk-config/

I used a local build, but action builds should also work i think: https://github.com/tompi/cheapinov2-zmk-config/actions/runs/11653138120

tompi, 2024-11-05:
So this is for a "true" wireless, with 2 mcus. Im gonna try to make a semi-wireless also, with cable+encoder and just one ble mcu.

jinchoung, 2024-11-05:
just tossing in my opinion but that 1 mcu solution is the sweet spot! that makes me excited! especially since there's a possibility i can just swap out the board???? that's probably not gonna work but one can dream!

azzamsa, 2024-11-05:
> especially since there's a possibility i can just swap out the board

Would you like to elaborate more on this?

SadWitchBoards, 2024-11-05:
I would definitely be in favor of dropping an encoder for two MCU. I really enjoy how Sadek incorporated his battery into the PCB by leaving a hole under the MCU on his bykeeb. Not sure if that's an option here but it allows for a thin case and a bigger battery than fits under the MCU in other builds. 

https://fingerpunch.xyz/wp-content/uploads/2024/07/PXL_20240809_022806306-scaled.jpg

SadWitchBoards, 2024-11-05:
P. S. Would anyone want to join a discord to discuss all things Cheapino/inexpensive ergo boards?

azzamsa, 2024-11-05:
@SadWitchBoards let's do. We can have it under *Symbiosys* in Absolem Club, under the channel name `Tompi`.
@tompi is that okay?

[image]

azzamsa, 2024-11-05:
> I would definitely be in favor of dropping an encoder for two MCU. I really enjoy how Sadek incorporated his battery into the PCB by leaving a hole under the MCU on his bykeeb. Not sure if that's an [...]

Or  you can put it under your palm à la Totem 
[image]

SadWitchBoards, 2024-11-05:
> @SadWitchBoards let's do. We can have it under _Symbiosys_ in Absolem Club, under the channel name `Tompi`. @tompi is that okay? [...]

Sounds great to me! I'm in that discord.

jinchoung, 2024-11-05:
> > especially since there's a possibility i can just swap out the board
> 
> Would you like to elaborate more on this?

oh, i just mean that i'm currently typing on my cheapino! it's all built and working great. what would be great for me is that for a wireless solution, i can just pull of the little RP2040 board controlling it now and adding in a replacement that gives me wireless.

azzamsa, 2024-11-08:
@SadWitchBoards ping me if you already requested `symbiosys/tompi` in Absolem Server.

SadWitchBoards, 2024-11-09:
@azzamsa I have not done that.

hasrthur, 2025-01-08:
Hi! Thank you for the great keyboard and your time to build such a beauty.
Is there any progress on wireless keyboard?

souljedi, 2025-04-16:
I love using my cheapino on a daily base. Any news on the nicenano version?

How is the seedunio prototype holding up after several months of usage?

tompi, 2025-05-05:
Hey, sorry for the late reply.

Seeduino is holding up just fine, but have to admit i dont bother using it with ble, i just have the cable in, right part is not wired, since the battery of that lasts very long...
So semi-wireless i guess :)

[image]

azzamsa, 2025-11-08:
Is Seeduino will be pushed here, or another repo? Can't wait 😍

tompi, 2025-11-10:
Hey @azzamsa , zmk code is here: https://github.com/tompi/zmk-config-cheapino-seeduinos
If you want to test, just clone the repo on github and trigger a build action, artifacts should be built.

But like i mentioned, battery life is not good(for the left part)

souljedi, 2025-11-13:
@tompi can you elaborate if you say battery life is not good? Maybe it is still okish, but not optimal. You mentioned that the left side is much shorter than right side?

My use case would be wired at desk and then occasionally going wireless for mobile usage. 

That said, I'm using the Cheapino daily now for more than one year and very very happy with it! 🎉

tompi, 2026-01-04:
@jinchoung check out this post: https://github.com/tompi/cheapino-zmk-config-single-nicenano/issues/1

jinchoung, 2026-01-06:
oh sweet! thanks tom, thanks abhishek!