tompi/cheapino #92: 4th row is ignored on custom layout (because of rotary encoder)

nohwnd, 2024-10-25:
Hello, 
first of all thanks for the project, it has been an amazing start for my journey into keyboards, and a great learning experience. 

I did not build it using your PCB because it has too few keys for me, and I also don't have access to cheap PCBs (or so I thought 2 weeks ago). 

Instead I am building this 3d printed model with hand wiring the keyboard.

https://www.printables.com/model/245613-the-ortholily-a-3d-printed-split-ortholinear-mecha/files

I've used the duplex matrix, that is used by cheapino to be able to connect the two halves via just 8 wires in ethernet cable, and use a single controller (to get more learning).  (This is an excellent article on the topic of [Japanese duplex matrix](https://kbd.news/The-Japanese-duplex-matrix-1391.html) if someone is a beginner like me.)

My problem is that 4th row is completely ignored (ZXCVB), but all other rows work, including 5th row. When I move the wire of 4th row to 3rd row all the keys work. I tried checking if this is a hardware problem, changing ports, changing rows, and nothing helped. 

I suspect this is a software problem and I solved the problem by adding an extra row that I don't connect, and setting all the keys to KC_NO. 

**I've eventually figured out that this is because of to the rotary encoder** (thanks to your [graphical debugging tool](https://tompi.github.io/cheapino/doc/troubleshooting/routing.html)). 

Do you think it would be worth it to point this out somewhere, even if just a comment in your source files?

nohwnd, 2024-10-25:
[image]
[image]

tompi, 2024-10-25:
Hey, cool keyboard!

Im a bit hesitant to call this a cheapino, since you completely build your own thing.

This thing has much more keys, the cheapino does not even have a 4th row, like you said, its only used for the encoder...

Im not sure how you wired it, but you should be aware that some encoders do not detent properly, and will continously output alternating symbols, which is why I dedicated an entire row to it(and made sure other columns going to the encoder has diodes pointing "into" the encoder, so it cant mess with anything but its own row...)

nohwnd, 2024-10-25:
> Im a bit hesitant to call this a cheapino, since you completely build your own thing.

I agree, but it is cheapino inspired for sure. 

It is wired as 6x5 matrix. The grey keys on the bottom + the "lock" key form a single row. I believe that is the maximum size that can fit to the 8 wires (5 rows + 6/2 cols = 8). 

For documentation sake:

This is my config, with the extra row, to work around the 4th row being for the encoder. Thanks for that info, I will try to find where in the code the encoder is considered and how I can remove or re-map it.

```json
{
    "manufacturer": "Thomas Haukland",
    "keyboard_name": "cheapino2",
    "maintainer": "tompi2",
    "bootloader": "rp2040",
    "diode_direction": "ROW2COL",
    "features": {
        "bootmagic": true,
        "command": false,
        "console": false,
        "extrakey": true,
        "mousekey": true,
        "nkro": true
    },
    // col 0 - orange
    // col 1 - orange white
    // col 2 - green

    // row 0 - blue
    // row 1 - brown
    // row 2 - brown white
    // row 3 - green white
    // row 4 - blue white
    "matrix_pins": {
        "cols": [
            "GP0", "GP0", "GP1", "GP1", "GP2",  "GP2"
        ],
        "rows": [
            "GP3", 
            "GP4", 
            "GP5", 

            "GP9",

            "GP6", 
            "GP7"
        ]
    },
    "processor": "RP2040",
    "url": "",
    "usb": {
        "device_version": "1.0.0",
        "pid": "0x0000",
        "vid": "0xFEE3",
        "force_nkro": true
    },
    "layouts": {
        "LAYOUT": {
            "layout": [
                
                { "matrix": [0, 0], "y": 0, "x": 0 },
                { "matrix": [0, 1], "y": 0, "x": 1 },
                { "matrix": [0, 2], "y": 0, "x": 2 },
                { "matrix": [0, 3], "y": 0, "x": 3 },
                { "matrix": [0, 4], "y": 0, "x": 4 },
                { "matrix": [0, 5], "y": 0, "x": 5 },

                { "matrix": [1, 0], "y": 1, "x": 0 },
                { "matrix": [1, 1], "y": 1, "x": 1 },
                { "matrix": [1, 2], "y": 1, "x": 2 },
                { "matrix": [1, 3], "y": 1, "x": 3 },
                { "matrix": [1, 4], "y": 1, "x": 4 },
                { "matrix": [1, 5], "y": 1, "x": 5 },

                { "matrix": [2, 0], "y": 2, "x": 0 },
                { "matrix": [2, 1], "y": 2, "x": 1 },
                { "matrix": [2, 2], "y": 2, "x": 2 },
                { "matrix": [2, 3], "y": 2, "x": 3 },
                { "matrix": [2, 4], "y": 2, "x": 4 },
                { "matrix": [2, 5], "y": 2, "x": 5 },

                { "matrix": [3, 0], "y": 3, "x": 0 },
                { "matrix": [3, 1], "y": 3, "x": 1 },
                { "matrix": [3, 2], "y": 3, "x": 2 },
                { "matrix": [3, 3], "y": 3, "x": 3 },
                { "matrix": [3, 4], "y": 3, "x": 4 },
                { "matrix": [3, 5], "y": 3, "x": 5 },

                { "matrix": [4, 0], "y": 4, "x": 0 },
                { "matrix": [4, 1], "y": 4, "x": 1 },
                { "matrix": [4, 2], "y": 4, "x": 2 },
                { "matrix": [4, 3], "y": 4, "x": 3 },
                { "matrix": [4, 4], "y": 4, "x": 4 },
                { "matrix": [4, 5], "y": 4, "x": 5 },

                { "matrix": [5, 0], "y": 5, "x": 0 },
                { "matrix": [5, 1], "y": 5, "x": 1 },
                { "matrix": [5, 2], "y": 5, "x": 2 },
                { "matrix": [5, 3], "y": 5, "x": 3 },
                { "matrix": [5, 4], "y": 5, "x": 4 },
                { "matrix": [5, 5], "y": 5, "x": 5 }

            ]
        }
    }
}

```

```
{
  "version": 1,
  "notes": "Cheapino default keymap",
  "documentation": "\"This file is a QMK Configurator export. You can import this at <https://config.qmk.fm>. It can also be used directly with QMK's source code.\n\nTo setup your QMK environment check out the tutorial: <https://docs.qmk.fm/#/newbs>\n\nYou can convert this file to a keymap.c using this command: `qmk json2c {keymap}`\n\nYou can compile this keymap using this command: `qmk compile {keymap}`\"\n",
  "keyboard": "bastardkb/skeletyl/blackpill",
  "keymap": "default",
  "layout": "LAYOUT",
  "layers": [
    [
      "KC_ESCAPE", "KC_1", "KC_2", "KC_3", "KC_4", "KC_5",
      "KC_TAB",    "KC_Q", "KC_W", "KC_E", "KC_R", "KC_T",
      "KC_LEFT_CTRL", "KC_A", "KC_S", "KC_D", "KC_F", "KC_G",
      "KC_NO", "KC_NO", "KC_NO", "KC_NO", "KC_NO","KC_NO",
      "KC_LEFT_SHIFT", "KC_Z", "KC_X", "KC_C", "KC_V", "KC_B",     
      "KC_NO", "KC_LEFT_ALT", "KC_LEFT_GUI", "KC_SPACE", "KC_SPACE", "KC_SPACE"
    ]
  ],
  "author": "thomas.haukland@gmail.com"
}

```

It is wired as duplex matrix where two columns are connected. The bottom keys are "shifted" to left so the matrix becomes more square. 

(colors in image don't represent the colors of the wire in the original photo, rows are green, columns are colored)
[image]

nohwnd, 2024-10-28:
[image]

My whole keyboard is now done. For my learning, can I ask you why did you choose to not share any rows in your design? I originally thought I will be able to connect the left side and the right side and use the same pins for the rows, but it did not work at all, and then I saw that you use separate rows (and of course columns) for the left and right part of the keyboard.

So now I am not sure if this is an optimization on your side, recommendation from qmk, or just an opinionated choice. 😊

Thank you!

tompi, 2024-10-30:
Hey, regarding separate rows/columns: there are still some ghosting issues, which are minimized when using different pins for right side. (the v1 cheapino shared rows across split, and therefor has some possible ghosts between both hands, which are much more likely to happen, than same hand ghosts...)

To test you own, try pressing two keys on the same row while pressing a third on the row below or above, on the "next" column. So 3 different columns, 2 rows. Not all combos produce ghosts, but on cheapinov2 I think I caught all of them in firmware.

nohwnd, 2024-10-30:
great thank you for the info. I will close this now since I never had any actual problem with cheapino 🙂

closed 2024-10-30 by nohwnd