tompi/cheapino #94: The creator of this open-source project has not been thanked sufficiently by me for their excellent contribution.

lightnin, 2024-11-02:
This is both well designed and well documented, and a wonderful example of an outstanding open source project.  My whole build went very smoothly thanks to your effort and your generosity, and of course the community of contributors who've chipped in / supported you.

A thousand thanks! Please close this whenever you like. But feel free to re-open it if you ever need a little encouragement for your efforts. They deserve to be acknowledged!

rfvizarra, 2024-11-02:
Completely agree! I have been following the activity for this project and Thomas is doing an amazing job. 
I haven't built a Cheapino... yet... but I would like to do it sometime.

tompi, 2024-11-02:
Thank you so much! I will keep the issue open a bit ;)

SadWitchBoards, 2024-11-02:
I 1000% agree. I have been always trying to spread the word to those who haven't used the board. Thomas, you make it so easy for beginners to build this board and give awesome instructions and support. I have been selling them for pretty much cost to those who don't want to build their own and everyone I speak with after loves their Cheapinos. I adore this little board and have gotten some of the best sounds of it the case of any of my ergo boards. Can't wait to see what you do next.

tompi, 2024-11-10:
OK, this was thanks enough ;)

closed 2024-11-10 by tompi