tompi/cheapino #95: Keyboard Layout Editor image for Miryoku-QWERTY (and other layouts?)

lightnin, 2024-11-04:
Hello good people -

I've decided to use my wonderful cheapino to try and learn Miryoku Qwerty. I have a flashed the excellent firmware provided.
However, I'm having difficulty figuring out a way to make a printable image of the keymap as a reference to help me learn (to be placed below my monitor).

www.keyboard-layout-editor.com seems like the right tool generate such an image, as it allows for representations of multiple layers on the same keys, which is nice for making a compact reference image similar to this one of [plain miryoku ](https://www.keyboard-layout-editor.com/##@@_f:4&w:12&h:0.5&d:true%3B&=%3Cb%3EMiryoku%3C%2F%2Fb%3E%3B&@_y:-0.5&x:2&c=%23c5c5c5&t=%23000000%0A%23ae0000%0A%23003ed6%0A%23068f00%0A%23ffffff%3B&=F%0AF8%0A*%0A8%0AExtra&_x:6%3B&=U%0A%0A%3Ci%20class%2F='fa%20fa-copy'%3E%3C%2F%2Fi%3E%0A%3Ci%20class%2F='fa%20fa-copy'%3E%3C%2F%2Fi%3E%0AExtra%3B&@_y:-0.87&x:1%3B&=W%0AF7%0A%2F&%0A7%0ATap&_x:1%3B&=P%0AF9%0A(%0A9%0ABase&_x:4%3B&=L%0A%0A%3Ci%20class%2F='fa%20fa-clipboard'%3E%3C%2F%2Fi%3E%0A%3Ci%20class%2F='fa%20fa-clipboard'%3E%3C%2F%2Fi%3E%0ABase&_x:1%3B&=Y%0A%0A%3Ci%20class%2F='kb%20kb-Scissors-2'%3E%3C%2F%2Fi%3E%0A%3Ci%20class%2F='kb%20kb-Scissors-2'%3E%3C%2F%2Fi%3E%0ATap%3B&@_y:-0.88&x:4&t=%23000000%0A%23ae0000%0A%23003ed6%0A%23068f00%3B&=B%0A%3Ci%20class%2F='kb%20kb-Unicode-PrintScreen-1'%3E%3C%2F%2Fi%3E%0A%7D%0A%5D&_x:2%3B&=J%0A%0A%3Ci%20class%2F='kb%20kb-Redo-1'%3E%3C%2F%2Fi%3E%0A%3Ci%20class%2F='kb%20kb-Redo-1'%3E%3C%2F%2Fi%3E%3B&@_y:-0.75&t=%23000000%0A%23ae0000%0A%23003ed6%0A%23068f00%0A%23ffffff%3B&=Q%0AF12%0A%7B%0A%5B%0ABoot&_x:10%3B&='%0A%0A%3Ci%20class%2F='kb%20kb-Undo-3'%3E%3C%2F%2Fi%3E%0A%3Ci%20class%2F='kb%20kb-Undo-3'%3E%3C%2F%2Fi%3E%0ABoot%3B&@_y:-0.5&x:2&t=%23000000%0A%23ae0000%0A%23003ed6%0A%23068f00%3B&=S%0AF5%0A%25%0A5%0A%3Ci%20class%2F='fa%20fa-angle-up'%3E%3C%2F%2Fi%3E&_x:6%3B&=E%0A%3Ci%20class%2F='kb%20kb-Multimedia-Volume-Down-1'%3E%3C%2F%2Fi%3E%0A%E2%AC%87%0A%E2%96%BC%0A%3Ci%20class%2F='fa%20fa-angle-up'%3E%3C%2F%2Fi%3E%3B&@_y:-0.8700000000000001&x:1%3B&=R%0AF4%0A$%0A4%0A%3Ci%20class%2F='mss%20mss-Unicode-Alternate-3'%3E%3C%2F%2Fi%3E&_x:1&n:true%3B&=T%0AF6%0A%5E%0A6%0A%3Ci%20class%2F='kb%20kb-Arrows-Up-Circle-Filled'%3E%3C%2F%2Fi%3E&_x:4&n:true%3B&=N%0A%3Ci%20class%2F='kb%20kb-Multimedia-Rewind-Start'%3E%3C%2F%2Fi%3E%0A%E2%AC%85%0A%E2%97%80%0A%3Ci%20class%2F='kb%20kb-Arrows-Up-Circle-Filled'%3E%3C%2F%2Fi%3E&_x:1%3B&=I%0A%3Ci%20class%2F='kb%20kb-Multimedia-Volume-Up-2'%3E%3C%2F%2Fi%3E%0A%E2%AC%86%0A%E2%96%B2%0A%3Ci%20class%2F='mss%20mss-Unicode-Alternate-3'%3E%3C%2F%2Fi%3E%3B&@_y:-0.8799999999999999&x:4%3B&=G%0A%3Ci%20class%2F='fa%20fa-lock'%3E%3C%2F%2Fi%3E%0A+%0A%2F=&_x:2&t=%23000000%0A%23ae0000%0A%0A%23068f00%3B&=M%0A%0A%0A%3Ci%20class%2F='kb%20kb-logo-amiga'%3E%3C%2F%2Fi%3E%3B&@_y:-0.75&t=%23000000%0A%23ae0000%0A%23003ed6%0A%23068f00%3B&=A%0AF11%0A%2F:%0A%2F%3B%0A%3Ci%20class%2F='kb%20kb-batman'%3E%3C%2F%2Fi%3E&_x:10%3B&=O%0A%3Ci%20class%2F='kb%20kb-Multimedia-FastForward-End'%3E%3C%2F%2Fi%3E%0A%E2%9E%A1%0A%E2%96%B6%0A%3Ci%20class%2F='kb%20kb-batman'%3E%3C%2F%2Fi%3E%3B&@_y:-0.5&x:2&t=%23000000%0A%23ae0000%0A%23003ed6%0A%23068f00%0A%23ffffff%3B&=C%0AF2%0A%2F@%0A2%0AOpp&_x:6%3B&=,%0A%0A%E2%87%A9%0A%3Ci%20class%2F='mss%20mss-Unicode-Page-Down-5'%3E%3C%2F%2Fi%3E%0AOpp%3B&@_y:-0.8700000000000001&x:1&t=%23000000%0A%23ae0000%0A%23003ed6%0A%23068f00%3B&=X%0AF1%0A!%0A1&_x:1&t=%23000000%0A%23ae0000%0A%23003ed6%0A%23068f00%0A%23ffffff%3B&=D%0AF3%0A%23%0A3%0ACur&_x:4%3B&=H%0A%0A%E2%87%A6%0A%3Ci%20class%2F='kb%20kb-Line-Start'%3E%3C%2F%2Fi%3E%0ACur&_x:1&t=%23000000%0A%23ae0000%0A%23003ed6%0A%23068f00%3B&=.%0A%0A%E2%87%A7%0A%3Ci%20class%2F='mss%20mss-Unicode-Page-Up-5'%3E%3C%2F%2Fi%3E%3B&@_y:-0.8799999999999999&x:4%3B&=V%0A%3Ci%20class%2F='kb%20kb-Unicode-Break-2'%3E%3C%2F%2Fi%3E%0A%7C%0A%5C&_x:2&t=%23000000%0A%23ae0000%0A%0A%23068f00%3B&=K%0A%0A%0A%3Ci%20class%2F='kb%20kb-Unicode-Insert-1'%3E%3C%2F%2Fi%3E%3B&@_y:-0.75&t=%23000000%0A%23ae0000%0A%23003ed6%0A%23068f00%0A%239f00d6%3B&=Z%0AF10%0A~%0A%60%0AButton&_x:10%3B&=%2F%2F%0A%0A%E2%87%A8%0A%3Ci%20class%2F='kb%20kb-Line-End'%3E%3C%2F%2Fi%3E%0AButton%3B&@_y:-0.33999999999999986&x:2.5&t=%23000000%0A%23ae0000%0A%23003ed6%0A%23068f00%0A%23ae0000%3B&=%3Ci%20class%2F='kb%20kb-Unicode-Escape-2'%3E%3C%2F%2Fi%3E%0A%3Ci%20class%2F='kb%20kb-Hamburger-Menu'%3E%3C%2F%2Fi%3E%0A(%0A.%0AMedia&_x:5%3B&=%3Ci%20class%2F='kb%20kb-Unicode-DeleteRight-Big'%3E%3C%2F%2Fi%3E%0A%3Ci%20class%2F='kb%20kb-Multimedia-Mute-1'%3E%3C%2F%2Fi%3E%0AM%0A%3Ci%20class%2F='kb%20kb-Unicode-DeleteRight-Big'%3E%3C%2F%2Fi%3E%0AFun%3B&@_y:-0.16000000000000014&c=%23888888&t=%23000000&f:2&w:12&d:true%3B&=%0A%3Ctt%3Egithub.com%2F%2Fmanna-harbour%2F%2Fmiryoku%2F%2F%3C%2F%2Ftt%3E%0A%0A%0A%0A%0A%3Ci%3EManna%20Harbour%3C%2F%2Fi%3E&_x:-2&c=%23cccccc&a:7&f:3&d:true%3B&=%3Cimg%20src%2F='https%2F:%2F%2F%2F%2Fraw.githubusercontent.com%2F%2Fmanna-harbour%2F%2Fmiryoku%2F%2Fmaster%2F%2Fdata%2F%2Flogos%2F%2Fmanna-harbour-boa-32.png'%3E&_d:true%3B&=%3Cimg%20src%2F='https%2F:%2F%2F%2F%2Fraw.githubusercontent.com%2F%2Fmanna-harbour%2F%2Fmiryoku%2F%2Fmaster%2F%2Fdata%2F%2Flogos%2F%2Fmiryoku-roa-32.png'%3E%3B&@_r:18&rx:3&ry:7.75&y:-4.09&x:-0.5&c=%23c5c5c5&t=%23000000%0A%23ae0000%0A%23003ed6%0A%23068f00%0A%23068f00&a:4&f:4&n:true%3B&=%3Ci%20class%2F='fa%20fa-space-shuttle'%3E%3C%2F%2Fi%3E%0A%3Ci%20class%2F='fa%20fa-space-shuttle'%3E%3C%2F%2Fi%3E%0A)%0A0%0ANav%3B&@_r:36&y:-1&x:-0.5&t=%23000000%0A%23ae0000%0A%23003ed6%0A%23068f00%0A%23003ed6%3B&=%3Ci%20class%2F='kb%20kb-Tab-1'%3E%3C%2F%2Fi%3E%0A%3Ci%20class%2F='kb%20kb-Tab-1'%3E%3C%2F%2Fi%3E%0A%2F_%0A-%0AMouse%3B&@_r:-36&rx:9&y:-4.09&x:-0.5%3B&=%3Ci%20class%2F='kb%20kb-Return-2'%3E%3C%2F%2Fi%3E%0A%3Ci%20class%2F='kb%20kb-Multimedia-Stop'%3E%3C%2F%2Fi%3E%0AR%0A%3Ci%20class%2F='kb%20kb-Return-2'%3E%3C%2F%2Fi%3E%0ASym%3B&@_r:-18&y:-1&x:-0.5&t=%23000000%0A%23ae0000%0A%23003ed6%0A%23068f00%0A%23068f00&n:true%3B&=%3Ci%20class%2F='kb%20kb-Unicode-BackSpace-DeleteLeft-Big'%3E%3C%2F%2Fi%3E%0A%3Ci%20class%2F='kb%20kb-Multimedia-Play-Pause'%3E%3C%2F%2Fi%3E%0AL%0A%3Ci%20class%2F='kb%20kb-Unicode-BackSpace-DeleteLeft-Big'%3E%3C%2F%2Fi%3E%0ANum).

However, I can see that the Cheapino Qwerty Miryoku firmware doesn't match this Miryoku in some ways. Can anyone recommend how to go about creating an image on www.keyboard-layout-editor.com based on the Miryoku Qwerty firmware provided for the Cheapino? 

If it's simple enough I'll be happy to do it for this layout and any other ones released, as I think it will be a useful reference for anyone trying to learn a new keymap.

Tak!

lightnin, 2024-11-05:
I just modified the layout above to make it QWERTY, easily enough. However, the Miryoku Qwerty available on the release page doesn't seem to match the layer behavior shown in the image. Perhaps there are differences because I'm on a Mac. But the media, num, and nav layers behave mostly as espected, just not copy paste. But the mouse and fun layers don't seem to behave quite right. 
[image]

tompi, 2024-11-09:
Hey, thats strange, I built the miryoku from the github repo...
Do you have an example of a key/sequence that is not as pictured?

lightnin, 2024-11-10:
Well, I think it's pilot error. I was confused about the symbols. It looks like everything is right save the (mouse or sim?) copy paste symbols on U I and O. Would I need to recompile with MIRYOKU_CLIPBOARD=MAC to fix that?

tompi, 2024-11-10:
Ah, Okay. Im not sure how that works, but since ctrl c/v is cmd c/v on Mac, it makes sense there might be some diff there...

Try recompiling. Let me know if you struggle setting it up, i remember there was an issue about this a while ago...

lightnin, 2024-11-10:
Yep, I went through quite an odyssey, but did manage to get things working with qmk. However, I think it makes the most sense to just use your stock firmware even if the copy / paste keys don't work, and try to stay close to Miryoku in hopes that it's an emerging standard. So the graphic I shared a few posts back should do the trick for me (and feel free to post it somewhere as a resource for others, if you think it would be helpful to people learning.) Thanks!

closed 2024-11-10 by lightnin