tompi/cheapino #96: any reason the cheapino would cause characters to be dropped with a wireless adapter?

jinchoung, 2024-11-09:
so let me preface this by saying that the cheapino on a wired connection with my computer is flawless. every key registers. 100% reliable. a joy to type on.

however, i was very impatient for a wireless solution (and also wanted to avoid a massive programming re-do if i need to swap the chip and reflash it using something else) so i got this: https://www.wscome.com/product/highly-recommendedwired-keyboard-converted-to-wireless-keyboard-diy-kit-v4-version-dual-interface-supports-keyboard-and-mouse-conversion-at-the-same-time/

i tried it with another standard keyboard and it works flawlessly as expected and hoped for.

however, with the cheapino, it doesn't get 100% of the keystrokes. it's like 98%. and as you can imagine, when you're used to touch typing quickly, dropped letters and spaces are absolutely a no go.

so my question is, is there any reason why the cheapino would cause the wireless transmitter to miss keys? it's definitely not the same keys so it's not a matter of just specific keys not working. it's just that occasionally, the transmitter just doesn't catch a key.

the device is purported to work with the vast majority of keyboards on the market and i can concur at least with an external thinkpad keyboard that i had. but yeah, cheapino is giving it problems and would love if there was an easy to implement solution for this.

thanks much!

jin

tompi, 2024-11-09:
The cheapino should work as any usb keyboard...

But just for testing, if you know you can get back your current setup easily, you might try flashing a few different layouts from the releasepage and see if it happens on all of them?

Do you know the make/brand of your dongle? Was a bit difficult to find any hits when googling...

You dont happen to have another qmk keyboard you could test with?

jinchoung, 2024-11-10:
hi thomas,

thanks again for fielding questions!

ok. i have a feeling that maybe the timing of super-key layers may play a part in keys getting dropped... will play with that when i feel like diving back into recompiling and stuff.

no, this is the only qmk keyboard that i have but i have a feeling that this would happen on most qmk custom keyboards (again, maybe as a result of timing of holding a key down).

have absolutely no idea who manufactures the receiver dongle. it's just black plastic with extremely sharp corners. but it works fine on "standard" keyboards. it's just my cheapino/qmk that gives it trouble. i'll try pinging the maker and see if he has any thoughts too.

thanks again thomas.

jin

tompi, 2024-11-30:
Hey, just a thought, there is this guy having a lag problem on mac login screen with x86, this buid seems to solve it:
https://github.com/user-attachments/files/17879491/cheapino_vial.uf2.zip

Could you test if it helps with your problem(turns off nkro and mouse keys..)

jinchoung, 2024-11-30:
hi thomas,

oh wow, thanks for remembering and this heads up! will check it out asap
and will report back with the results. thanks again!

jin

jinchoung, 2024-11-30:
oh actually, i don't remember the .uf2 file that i'm currently running but
i'd hate to lose it in case this new one doesn't work out. is there a way
for me to make a back up of what's currently on the keyboard?

thanks!

jin

tompi, 2024-12-02:
Hey, I think you need the desktop version of Vial.

There you should be able to do "File->save"
The you can load that file later after you flash something