tompi/cheapino #97: Which RJ45 socket should I pick?

azzamsa, 2024-11-09:
Hi 👋🏽

My local shop has these options:

- **Type 58**: 8P8C, Metal Shield, Horizontal 90°, No LED
- **Type 56**: 8P8C, Metal Shield, Horizontal 90°, with 2 LEDs
- ~~**Type 5224**: 8P8C, Metal Shield, Vertical 180°, No LED~~ (vertical!)

Would you recommend going with Type 58 or Type 56? Or would either work just as well?

I also noticed each socket has 8 pins, but the PCB has 9 holes. Could the extra hole be for a reversible design?

Thanks!

[image]

tompi, 2024-11-09:
Hey. You want type 58. Type 56 has a different layout it seems. It also has some extra pins for the port leds, which you don't want.

Go with type 58.

And yeah, the ninth hole is for reversibility. Just align connector with outline on the side you are mounting it on and you should be fine.

azzamsa, 2024-11-09:
> Go with type 58

Okay!

> And yeah, the ninth hole is for reversibility.

Oh, I see.

Thanks a lot! 🥞

closed 2024-11-09 by azzamsa