tompi/cheapino #98: Modifying the heat inserts on 3D case

azzamsa, 2024-11-10:
Hi,

This is the only local shop here that sells heat inserts with flanges. Sadly, the length is limited, and I have to modify the OpenSCAD file. I chose the one with the flange because people said it is cleaner to work with a soldering iron.

[image]

Do you have any recommendations on which one I have to pick?

I think M2 x 4 (inner diameter) x 4 (length) is a good candidate here.

---

I tried opening the file, but I'm still struggling to find the code to change to M2 x 4 (inner diameter) x 4 length.
I used the development version (appimage), but the rendering time is slow (on the powerful machine) and the rendered image is distorted, even though I have followed your guide.

I think I will modify using FreeCAD. Rotating the rendered image in OpenSCAD is super slow. I can't even zoom properly.

[image]

tompi, 2024-11-10:
Hey, did you read the readme file that is in the case folder? There are some options to drastically speed up rendering.

Also, if you render it using F6, it will look smoother and be quicker to interact with.

To change heat inserts, you will want to tweak the "mounting_hole_insert()" function in "modules.scad"

Check alignement of holes carefully after changing them!

tompi, 2024-11-10:
I THINK it should be enough to change line 383 to:

cylinder(h=4, r=1.9);

azzamsa, 2024-11-10:
> Hey, did you read the readme file that is in the case folder? There are some options to drastically speed up rendering.

Ah, I only doing the 3 million, and somehow skip the manifold. It is not in `Feature` tab anymore. The docs doesn't say anything about this and google doesn't help. So I went to the IRC and they point me to https://fosstodon.org/@OpenSCAD/113256867413539398. I will make a related PR to Cheapino doc.

Btw, the original measurements are:

```
// M2 heat inserts. 3mm outer diameter.  4mm length
cylinder(h=4, r=1.4); 
```

I wanted to use `M2 heat inserts. 4mm outer diameter.  4mm legth`, how do you come up with `1.9` where do you get this number? 

```diff
@@ -380,7 +380,7 @@ module mounting_hole_insert() {
     color("#bcbd22")
     translate([45.85, -23, 0]) {
       translate([0,0,4.1])
-      cylinder(h=4, r=1.4);
+      cylinder(h=4, r=1.9);
     }
 }
```

https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Primitive_Solids#cylinder

closed 2024-12-29 by tompi