tompi/cheapino #102: Guide for installing Xiao BLE and the on/off switch on cheapinov2

relmoo1220, 2024-11-19:
Hi Thomas,

Thank you so much for the CheapinoV2. I’ve been really enjoying typing on it!

I’ve read through some of the discussions about the wireless version and understand that it’s not fully ready yet. When it becomes available, could you please create a guide detailing which specific Xiao BLE to purchase, how to swap it out, and where to install the on/off switch on the left/right side of the PCB? I’m looking forward to building a set of wireless CheapinoV2. 

Thanks again for all your hard work and support!

tompi, 2024-11-19:
Hey, as of now, it works fine, but it doesnt go into deep sleep, so battery does not last many days if you dont turn it off...
Im not sure im able to solve this without changes to the pcb...(a rev 3...)

If you want to try this, it really isnt very complicated, you can just mount the xiao at the "bottom", and flip it on the right side.

ZMK support is here: https://github.com/tompi/cheapinov2-zmk-config 

But like I said, this isnt very polished...
(btw, I also feel like zmk is somewhat lacking... Eg the new feature zmk studio is nowhere near vial...)

[image]

relmoo1220, 2024-11-21:
Thank you for the photo reference! I don't think I would mind charging it once every few days, so I'll just get the Xiao BLE and battery to try it out. May I know what battery you are using in the photo? Seems a little big in size so maybe I'll get something similar but less battery capacity and smaller in size.

tompi, 2024-11-21:
The battery is 3.7V 200mAh.

Its way too big. Im waiting for a pair of 1cm x 2cm x 4mm, also 200mAh, which I hope will fit under the mcus.

Batteries from ali seems to have a longer shipping time...

If you want to go cheap you can probably also go for a pair of nice nano clones(1/3 of the price of the seeduino...), but I have not made fw for a true split for this yet(i have it on the top set in the pic with zmk and non-split..). The downside of this is its a bit longer, so it sticks out a bit in front...

Gotta say the wireless thing is growing on me, been using it for work for a week, and no problems...

relmoo1220, 2024-11-22:
Thanks for the clarification on the battery sizing!

Do you have the zmk config running the miryoku layout?

Also, when connecting to MacOS do you select the ANSI layout?

tompi, 2024-11-22:
Im running my own keymap(resembles colemak dh miryoku, but not quite the same...)...

For miryoku you would need to look into building it out-of-tree, not quite sure how that works, but probably not very difficult.

Im using us-international on mac, so I can get norwegian characters with alt-gr combinations.

relmoo1220, 2024-11-22:
For the Xiao BLE is it able to just load the miryoku zmk config they provided?

tompi, 2024-11-22:
I dont know how it works... but you need to combine 3 repos somehow: zmk, cheapino-zmk and zmk-miryoku.

HOW to do that, I dunno, sorry... If you figure it out, let me know, and I will post it to my pages!
Probably the guys at zmk discord can give a nice recipe!

relmoo1220, 2024-11-22:
Ahhh i see thanks a lot!

I guess just one last thing would be do you have a photo reference of the different layers for your zmk config?

That would be much easier to visualize the key mappings.

Thanks for the fast replies!

tompi, 2024-11-22:
Sure, the actual zmk keymap is here:
https://github.com/tompi/cheapinov2-zmk-config/blob/main/boards/shields/cheapinov2/cheapinov2.keymap

To get a visual rep, download this file:
https://github.com/tompi/qmk_firmware/blob/cheapinov2/keyboards/cheapino/keymaps/tompi/keymap.json

And upload it here:
https://config.qmk.fm/

The zmk keymap is missing some special qmk codes, like led control and mouse movement(which is missing from zmk still...), but the important stuff should be there

relmoo1220, 2024-11-22:
Thank you very much for your help, I will proceed to mark this as closed :)

closed 2024-11-22 by relmoo1220

tompi, 2024-11-22:
Send me a note if you build the wireless version, im very curious about your experience!

I can probably help you if you get stuck trying to build miryoku.

P.S. Battery drains slower than I feared, seems like less than 5% per day if I remember to turn it off at end of day.

tompi, 2024-11-22:
And if you take some pictures or have ideas about mounting, I will be very happy to incorporate them into a guide!

tompi, 2024-11-22:
Yep. That's the one. [image]