tompi/cheapino #103: Nothing detected

uivmc, 2024-11-20:
I finished building cheapino, flashed the uf2, got the LED to turn on, and detected it on vial.rock. But the matrix test gave an error and none of the keys worked.
Then I noticed I needed to solder the legs to the MCU (I used socketed MCU).
After I did that the MCU LED is off, vial.rock can not detect it anymore, and still none of the keys work.
I've reflashed the uf2 (the MCU can still show up as a USB drive. I notice this time the uf2 file disappears after I drag it on, originally it would stay there), and reset my computer. I've used a multimeter to check the legs at the top of the MCU are electrically connected to the solder at the bottom of the PCB. I've shorted two pins of a switch and still no keypress. Nothing has worked.
What should I do?

tompi, 2024-11-20:
Hi, could you post a picture of the mcu? (maybe some solder bridges?)

If you test with mcu out of socket, does the led light up?

uivmc, 2024-11-20:
Hi.
[image]
[image]
Even out of socket no light.

uivmc, 2024-11-22:
I was wondering if I damaged the MCU.
But I was able to do the blinking led demo via Arduino IDE from this video
https://www.youtube.com/watch?v=eTf5z5lcWhc
so I know it is still functioning.

azzamsa, 2024-11-23:
I have a similar issue (nothing detected), and it was a wrong firmware #104. I am using v1 for v2 board.

uivmc, 2024-11-23:
I had made sure to download v2 uf2. And my PCB says v2 so they match.
In your issue you mention a cheapino and cheapinov2 branch. But neither exist to me. I only see a master and fix/case-tolerances branch. Do you mean tags?
I tried the v1 uf2 just in case and interestingly vial can detect and connect with that. The keys give the wrong outputs though, which matches what the troubleshooting guide predicts for incorrect version.
I also noticed that the v2 uf2 file was updated 4 days ago, which is 1 day before I downloaded it...
@tompi  Can you provide the old version of the v2 uf2 file? I want to check if that works.

tompi, 2024-11-23:
Hey, you are correct, I changed the default brightness of the LED(turned it way down...), and I probably should up the release version, but until now I kind of feel like the gerber is the versioned artifact... but i will take care. and not change releases in the future.

The attached uf2 is from july, please let me know if it works!

Could you maybe try with some other usb cables?
I know there are issues with "too good" cables and the cheapino, so if you have a usb2 cable, it would be interesting to hear if that works better.

[cheapino_vial.uf2.zip](https://github.com/user-attachments/files/17879288/cheapino_vial.uf2.zip)

tompi, 2024-12-29:
Hey, sorry I forgot to reply to this one... Did you figure out what was wrong?

If not, is vial recognizing your out of socket rp2040 if you flash it with any of these uf2 rp2040 files? https://okin.gitlab.io/vial-qmk-firmwares/

uivmc, 2025-01-02:
I did not figure out what was wrong.
Vial detects the out of socket (and also in socket) rp2040 when I flash asmodus_plaquead_rgb_vial.uf2 onto it.

tompi, 2025-01-02:
Which host is Are you using? And did you try on another machine or with a docking station?

uivmc, 2025-01-02:
I used the online vial. https://vial.rocks/
In the past I have tried: on another machine, with usb-hub, and without usb-hub

uivmc, 2025-04-14:
Finally got around to ordering some more MCUs and soldering them. That fixed the problem.
Strange that the old MCU can still run the LED uf2s with no problem.
Unfortunately, I encountered problems when disconnect and connect the magnetic USB. Power surge warnings, or keypresses not working. It would sporadically work again at times. Until eventually one time I disconnect and my computer went black or sleep mode or something briefly, the MCU lights went out, and now I can not even put it into dfu mode anymore.
Luckily I have one more spare MCU. I'll stick with a non-magnetic USB and see if other problems arise.
This issue was resolved at least, old MCU having problems, so I'll close this.
A suggestion. Originally when I clicked the MCU aliexpress link it only let me order 1, due to the newcomer discount or something. Which is why I originally didn't order spares. After you create an account and sign in though, you can increase the quantity although without the discount. Could mention this in the ordering guide when you are recommending people to order 2 MCUs, so they know how to.

closed 2025-04-14 by uivmc