tompi/cheapino #105: 2 Keys keys stopped working. Any ideas?

ReedMerrill, 2024-11-23:
I built my cheapino about a month ago and have used it daily at work for a couple of weeks. Then, yesterday, in the middle of work my "x" and "v" keys stopped working. I swapped the switches and those keys still didn't work, so at least I ruled out an issue with the them. I feel like it might come down to a hardware issue, but I'm not sure what, other than maybe a cold joint that I didn't know about. Any ideas what I should do next to debug the issue?

tompi, 2024-11-23:
Hey. This could be a problem with the diodes for these keys(broken diodes or something with the solder joint).

If you have a multimeter, you can test for continuity across the diode. 

You can also test shorting the switch "before" the diode. 

Maybe start by inspecting the diodes close to the switches not working

tompi, 2024-12-29:
Closing after no response, feel free to reopen if you want to continue debugging...

closed 2024-12-29 by tompi