tompi/cheapino #109: Possible to make the case parametric to allow for different sized encoders and RJ45 sockets?

glennmca-dev, 2024-11-28:
I recently started building my second cheapino for work. i used the ordering guide to get some encoders, but the encoders i recieved seem to be a different size from those that the case is designed for (the one i bought direct from tompi)

I had to cut the case to allow the new rj45 and encoder to fit. the new encoders from the order guide link are:
13.8mm from 3pin to 2pin side (north to south when installed on the pcb) and 11.9mm east to west

The encoder the case is designed for is 	11.9/12mm and seems to be perfectly square.

The RJ45 was my mistake, i didn't look at where the support pins were in the photos, they're near the pcb pins at the back on this model, but i cut them off and it works

tompi, 2024-11-29:
Hey. Thank you, i will update the guide. Didnt know there was different
sizes in the links:(

***@***.***, 90786985

glennmca-dev, 2024-11-29:
There doesn't seem to be any option for different sizes, just the 15/20mm encoder shaft.
I think maybe the seller changed supplier and they just happen to be a different size.

tompi, 2024-12-29:
Hey, I think the above PR should fix the encoder cutout size. Feel free to reopen if you disagree!

glennmca-dev, 2024-12-29:
Hey, Yeah i think it should fix, it may still be a little tight as only 1.6mm clearance was added, where the encoder i bought is 1.8/1.9 mm larger in the Y direction, but it's nothing a little bit of sanding or filing wont fix. 0.2/3 of a mm shouldn't be an issue 😄