tompi/cheapino #110: Having minor issues with the right home row

JuneAtalanta, 2024-12-10:
My first DiY and I am in love. Super easy to do and it even (almost) works! As in the title though, my right side home row is not working. I am reasonably confident that the solder joints on the RJ45 and control board are good, and I also tested with a few cables. Not sure what else to try though?

tompi, 2024-12-11:
Hi, the interactive schematics could be could for this case.

https://tompi.github.io/cheapino/doc/troubleshooting/routing.html

Here are the points you need to check carefully:
[image]

JuneAtalanta, 2024-12-12:
Huh, honestly a bit more lost as the thumb cluster button for this group works, but not the other keys.

That being said, absolutely love this. Great documentation tool ^-^

tompi, 2024-12-12:
So middle thumb on right side works? Thats strange, since it is on the middle row wire...

Anyway, Are you using the vial firmware? Could you check the test page there and see which keys Are missing? That will rule out any Key map errors...

tompi, 2024-12-12:
One other possibility is that you oriented all the diodes for the right middle row in the wrong direction, seems like a strange coincidence though...

JuneAtalanta, 2024-12-16:
Good morning, and happy Monday. I rechecked everything with no luck, and ended up just putting together one of the spare boards that I had ordered. Thankfully this worked! Must have been a bad board of something? Thankyou for all of the help!