tompi/cheapino #111: Right side not working & single key

zshzebra, 2024-12-25:
My key where F is on QWERTY places it isn't working, I will check tomorrow when I wake up, but another issue is that my whole right side apart from the encoder doesn't work. I used the white angled RJ45 cable lots of others use, and another one lying around and still the encoder worked but no keys.

Could someone let me know what debugging steps to do.

For the single key I think a replacement diode or switch will fix it, but the right side is stumping me.

tompi, 2024-12-25:
Hey!

One key missing is probably broken diode or diode in wrong direction or malfunctioning switch. 

Could you post a picture of your right side? Did you remember to bridge the solder pads? (On the right side only)

zshzebra, 2024-12-25:
Thanks for the prompt reply.

In my haste to get to bed I forgot to mention that the right side on BOTH keyboards I built weren't working apart from the encoder. I also had some issues with the left side on the other keyboard with a row not working, but a diode orientation or similar is likely to blame.

I will attach some photos soon.

zshzebra, 2024-12-26:
It's ok, on the right side I put the switches in the wrong way 😂.

closed 2024-12-26 by zshzebra