tompi/cheapino #113: V2 Case thin walls

ktowen, 2024-12-28:
I sent the case to jlc3dp.com to 3d print and they sent me an email after reviewing the files saying that some walls are too thin and could break during printing. I don't have any experience with 3d printing, is the risk acceptable or is this something that can be fixed?

This is the email they sent:

> the wall thickness of those indicated red area is too thin, which has high risks of loss, deformation ,crack, or damage during the printing process.
> 
> Could you please confirm if the risks are acceptable for you?
> 
> If not, could you please kindly increase it to be at least 0.8 mm to proceed? 1.5 mm will be better.Then we will activate the ' replace file' button for you.
> 
> [image]

glennmca-dev, 2024-12-29:
The walls are thin, but i've printed the case twice. Once on a badly calibrated ender 5 and once on a bambulab A1. It's been fine both times, Probably worth the risk

tompi, 2024-12-29:
Several people have printed it, but yeah, the are thin. 

I hope you print using PLA?

I printed a few skeletyls at jlcpcb, and also got that warning then...

ktowen, 2024-12-29:
Thank you all for the clarification. I set it to print in resin, let's see if that works well

closed 2024-12-29 by ktowen

tompi, 2024-12-29:
Should be ok, but be very careful when you mount the melt inserts. Resin kind of degrades in stead of melting.

schuay, 2025-07-10:
> Thank you all for the clarification. I set it to print in resin, let's see if that works well

@ktowen How did it turn out? I'm at the same step right now. 

FYI it's not possible to select PLA since the model is too small. "The minimum build size for PLA is 3.00x3.00x1.00cm, please change the material." Do you perhaps have other material suggestions?