tompi/cheapino #115: Media keys sent as different keyboard

imavlr, 2024-12-29:
I'm running the vial firmware on the cheapino V2 on a linux system.

The first thing I noticed after switching from my old keyboard is that I can no longer use any modifiers with the media keys, XF86VolumeMute specifically.

After some troubleshooting, it seems like the XF86VolumeMute gets sent as if it is from a separate keyboard, since `wev` prints the usual info it does when getting input from keyboard that is different from its last input.

Initially I thought it might be because I was using the encoder for the mute button, but even after mapping KC_MUTE to a normal key, I see this behavior.

I don't know enough about QMK or vial to tell if this is for sure what is happening, but it looks like it to me.

tompi, 2024-12-29:
Sorry, I have no idea whats going on here... Only thing I would try is using the KC_KB_MUTE keycode, and see if thats better?(I dont know what the difference is, but seems to be a different code sent...)

tompi, 2024-12-29:
Asked on the qmk discord and got this answer:
[image]

Tried to compile VIAL now with shared_ep, coud you try this fw and see if it works?
(ps remember to backup your keymap first!)

[cheapino_vial_shared_ep.uf2.zip](https://github.com/user-attachments/files/18269809/cheapino_vial_shared_ep.uf2.zip)

imavlr, 2024-12-29:
Wow, didn't expect a solution so fast, but everything works as expected now.

Thank you so much!

closed 2024-12-29 by imavlr