tompi/cheapino #116: Add a LICENSE file to the repo

hoijui, 2025-01-01:
... in case you want to have it as Open Source.

Without this, anyone building or modifying your design, could potentially be sued by you in the future, from a legal standpoint.

tompi, 2025-01-21:
Hey, an oversight by me, never really cared :). Does this work?

closed 2025-01-21 by tompi

hoijui, 2025-01-22:
That is much better already, thanks! :-)
I would recommend to put this file:
<https://creativecommons.org/licenses/by/4.0/legalcode.txt>
in the repo root with the name `LICENSE.txt`.
That way, GitHub and any other software scanning for licenses, will definitely be able to pick it up and detect it as `CC-BY-4.0`, and show that in different parts of its interface. It is also what humans will look for first.

commit 3c4f6ed5 (2025-01-21): Add license: fix #116