tompi/cheapino #119: Help with miryoku with VI

zumosik, 2025-01-06:
I really want to use miryoku on my cheapino, but with VI option (so i have hjkl). I have tried:
```shell
qmk flash -kb cheapino -km manna-harbour_miryoku \
> -e MIRYOKU_NAV=VI
```
but got default miryoku (with neio instead of hjkl).

- Command I got from [here](https://github.com/manna-harbour/miryoku_qmk/blob/miryoku/users/manna-harbour_miryoku/readme.org#qmk), but for cheapino, instead of ```crkbd```. 
- VI option from [here](https://github.com/manna-harbour/miryoku/tree/master/docs/reference#nav-1) 

So I would appreciate any help on setting miryoku with VI option.

tompi, 2025-01-06:
Hi. This is deep into Miryoku details... Do you have any idea about this @manna-harbour ?

zumosik, 2025-01-07:
Couldn't figure it out, so I just did it using qmk configurator.

closed 2025-01-07 by zumosik

manna-harbour, 2025-02-12:
Other than the suspect line continuation, what you had looks correct, i.e. ```qmk flash -kb cheapino -km manna-harbour_miryoku -e MIRYOKU_NAV=VI``` should work. Alternatively, you can set options in `custom_rules.mk` e.g. ```MIRYOKU_NAV=VI```.