tompi/cheapino #120: Some Keys Not Working

ecunningham418, 2025-01-11:
Hello, 

Excuse me if these seem like silly questions as this is the first soldering/electronics project that I've done.

I initially had a problem where one of the switches was not working but then found out that I had missed a diode during assembly. Upon soldering in the diode, however, the keys in that column also stopped working.

[image]

This was the initial key that did not work before putting in the diode.

[image]

Afterwards, all of these keys did not work. I have been trying to follow the troubleshooting guide and I am a little lost (such as how to short the switch before the diode), but to me it seems like the issue is one with the MCU? But I am not entirely sure due to how well it was working before I put the diode in.

[image]
[image]
[image]
[image]
[image]
[image]

Once again please excuse me if these sound like silly questions, I am still not entirely sure on what I'm doing 😅

ecunningham418, 2025-01-16:
Update: I got it to work! Issue was the MCU, I replaced it and it is now working great! Typing this out with the completed board.

closed 2025-01-16 by ecunningham418

tompi, 2025-01-16:
Hey. Great that you figured it out. It seems like some of these cheap rp2040-zeros break easily:( had some reports like this, but then again, i only hear of the ones that to bad :)