tompi/cheapino #121: Name of PCB (very silly question)

antiqstav, 2025-01-12:
Hey all, I am a new person to keyboards. My friend told me that my hands always tilt when I type on a keyboard, so after a bit of searching on the internet, here I am.

I was looking through the Cheapino and I was wondering if there was some way to change the name on the PCB to some kind of nice geographical feature, like an Apple update name. The main thing is to have the PCB not show Cheapino in an Impact font.

antiqstav, 2025-01-15:
Could I get some help on this? I can change it by myself if someone has a tutorial on it.

tompi, 2025-01-15:
Hi. This should not be to difficult. You need to install the program kicad and open Io the cheapino files in there and change them. 

Find a kicad tutorial on YouTube.

antiqstav, 2025-01-15:
Ok, thanks. I'll find out how to do that, but I'm not too well-versed in this kind of thing. Is there anyway that you could change the name and possibly the grb file to me? (pretty please with a cherry on top)

tompi, 2025-01-17:
Hehe, sorry, but this will never make it to my TODO-list ;) Good luck.

closed 2025-01-17 by tompi