tompi/cheapino #122: Help Needed: Issues with Key Detection on Cheapino Build

sennecools, 2025-01-14:
Hi everyone,

I recently built my first Cheapino V2 keyboard, and I’m running into some issues. Some keys (highlighted in red) work perfectly, while others (highlighted in blue) only work sometimes or don’t register at all. This is my first time soldering or working with PCBs, so I’m not sure if the problem is related to my soldering or something else.

I noticed that I installed the diodes on different sides of the PCB: on one half, they’re on the top, and on the other half, they’re on the bottom. However, the orientation is correct since the lines on the diodes match the square markings on the PCB.

Additionally, I used the legs of the diodes to connect the MCU, but the connection feels very loose, and I have to press on the MCU for it to connect properly to the board.

Do the diodes being on different sides matter as long as they’re oriented correctly? Could weak solder joints be causing the inconsistent key detection? And how can I make the MCU connection more secure?

Any advice would be appreciated. Thanks in advance!
[image]
[image]
[image]
[image]
[image]

sennecools, 2025-01-14:
It show up in vial and I have tried different cable, windows and mac.

tompi, 2025-01-14:
Hi! I see you have the slightly more expensive socket, which is mesket for thicker legs. I think the pin headers that comes with the MCU Are suitable(you can loosen them from the plastic).

I specified this in the ordering guide, but should probably mention it in the build guide as well. 

If you want to fix it as is, you can figure out which pins to resolder/reflow using this tool: https://tompi.github.io/cheapino/doc/troubleshooting/routing.html

sennecools, 2025-01-15:
Alright I tried using header pins that came with the mcu on my second mcu but now it wont even power. There is no light or anything tested it before and it was working and detected in vial now it sometimes gives the windows sound but not showing up in vial or as a drive. I have tried multiple cables. Did I maybe overheat the board while soledring or something?

[image]

tompi, 2025-01-15:
Does it work out of socket? Also worth trying without the magnetic connector...

tompi, 2025-01-15:
But yeah, if it does not show up as thumb drive when you reset it out of socket then something is very wrong :(

tompi, 2025-01-15:
Just check carefully for shorts. If there Are not any shorts, then my guess its unfixable :(

sennecools, 2025-01-16:
Yeah fried the second mcu as well I think it keeps disconnecting and flashing itself. Guess I need to order some new parts and work on my soldering skills, btw the link for the diodes seems to not be available anymore.

tompi, 2025-01-16:
Just be careful with your magnetic usb connector. That might be the culprit...

sennecools, 2025-01-17:
Ordered some new mcu since I tried everything with the ones I had. Any tips for soldering the mcu I used 400 celcius but maybe that's too hot, also do you wait inbetween each pin for the board to cool down a bit?

tompi, 2025-01-17:
I had to check, and mine is set to 380c. I think 400 should be fine. 

Im guessing the important thing is how long you spend at each pin. Dont use more than a fee seconds. 

But Im no Expert at all(the electrical engineer at work keep remarking i need to step up my soldering game...)

sennecools, 2025-01-17:
I'll try to be extra carefull next time then, thank you for all the help :)

sennecools, 2025-01-22:
I have no idea what I'm doing wrong tried with a new board and I always end up with 1 of the pins connecting when I test with a multimeter but there is nothing I can see on the board.

tompi, 2025-01-22:
What do you mean one of the pins connected?

sennecools, 2025-01-23:
When I put the multimeter on two pins next to each other it beeps so they are connected somewhere. But even when I look closely at the board I cant see any solder inbetween them. and I watched out this time only a couple second per pin. waited for the board to cool off a bit inbetween pins.

Thearne4, 2025-03-17:
the pins connect inside of the MCU (with some measurable resistance between them). try detecting again without the MCU installed 😉

tompi, 2025-08-22:
Closing after no activity. Feel free to re-open if this is still an issue...

closed 2025-08-22 by tompi