tompi/cheapino #123: Help to source simplest plates, please?

dansteeves68, 2025-01-21:
After using Ergodox for a few years and ignoring all the extra keys, I ordered everything to make my first Cheapino. I would like to use a plate (or inexpensive case?), but am at a loss for sourcing it as JLPCB says the .stl files are too small for 3d printing and I don't see how to ask for them laser cut out of PCB material. Can someone provide a bit more detail? Thanks.

tompi, 2025-01-21:
I think you can print them at jlc? Hmm, I just checked and FDM is not possible, but resin and nylon both should work.
(need to be careful with the resin when melting in the nuts...)

But really, if you know somebody, or have access to a makerlab, like at a library, it doesnt take long, and does not use much filament...

closed 2025-04-21 by tompi

dansteeves68, 2025-04-21:
I did finally figure out how to order them 3d printed from JLC.