tompi/cheapino #124: Third row and encoder not working on the right side after desoldering

GabrielCanaple, 2025-01-28:
Hello!

# The problem

First, thank you very much for your project, it's awesome ! I already got one cheapino working well with soldered switches, and it's a pleasure to use it.

However, for my second build, I got a problem. I soldered the encoder upside down (too eager to finish soldering and start using it !).
One thing to note though is that all the keys of the right side (and the encoder) were working fine, even with the encoder upside down.

So, I had to desolder it (which broke the encoder) to put a new one instead. And now, the third row and the encoder are not working.

#  What I tried

- I used my other (working) cheapino to make sure it's not caused by the left side. It is indeed not the left side.
- I re-soldered the diodes for the whole row, but it didn't solve the problem (see photos to make sure my solders are not too crappy)
- Re-soldered the corresponding pins on the RJ45 port. No success.

# My theories

1) When de-soldering, I burnt the PCB and the underlying copper wires. See photos, there is a small scratch on the other side of the encoder, but I am not sure it's the culprit (that's why I'm posting).
2) I did a bad solder somewhere and am not able to re-solder it properly.

Do you have any idea about what's wrong here ? Thank you !

[image]
[image]

tompi, 2025-01-28:
Hey. If you have a multimeter you can probably track this down. 

So you attached your working right side to the new left and everything worked?

Check out the interactive schematic: https://tompi.github.io/cheapino/doc/troubleshooting/routing.html

And measure continuity along the non working traces

GabrielCanaple, 2025-01-29:
Thank you for your answer. 
Yes, I did all the combinations (old left with new right, old right with new left) to make sure that the problem was with the new right.

I do not have a multimeter at the moment, and will need a few weeks to get my hands on mine (not available atm). I will update when I do.

Although, I do think that it is weird because this row and the encoder of the new right were working before desoldering the encoder. Could this be due to overheating the PCB or something like that ?

tompi, 2025-01-29:
Yeah, but third row is strange. Im more inclined to think something happened with the rj45 socket. 

Or a scratch in the pcb, severing third and fourth(encoder) row.

GabrielCanaple, 2025-03-09:
Hello again,
I finally managed to get my multimeter, and tested the continuity on the right side.
Your intuition was right, it was indeed the RJ45 socket (no continuity between the socket and the board). I replaced it, and it works perfect !
Thank you for your help.

closed 2025-03-09 by GabrielCanaple