tompi/cheapino #127: Diode quick question

glebino, 2025-02-06:
Is it possible to mix different types of diodes? I was out of long leggy ones and tried to substitute one with small surface diode, but it's not working. Is it because you must have one type of them on your pcb? Thanks

glebino, 2025-02-07:
Wrong diode placement + mcu was broken. It is certainly possible to use those diodes together

closed 2025-02-07 by glebino

tompi, 2025-02-07:
Sorry, i missed this one... glad you figured it out! Should be no problem to mix.

glebino, 2025-02-07:
@tompi thanks! In the process of figuring this out, 3 mcus died 😥. Very fragile those ones, isn't it?

tompi, 2025-02-07:
I think you might have been unlucky... made around 15 boards without any problems. 

Where did you buy them? Link?