tompi/cheapino #129: Slave half just started acting up.

TheTechRun, 2025-03-04:
As the title states my slave half just randomly started acting up. 

Issues:
The middle row on the right side (slave) outputs random characters. In qwerty  "H" outputs "ujm", "J" outputs "ujm". "K' outputs nothing, "L" outputs "ol.", and ";" outputs nothing.

What I have tried:
Using a different ethernet cord.
Using a different usb-C cable.
Connecting to a different computer.
Reflashing the firmware. 

Any ideas?

tompi, 2025-03-04:
Hey. If you Check the trouble shooting guide, there is an interactive schematics, which should help you track down points of interest depending on keys presses. 

I would use this and then Check the indicated paths. 

My guess is you have a wire stuck somewhere that is shorting stuff...

TheTechRun, 2025-03-04:
I bought this prebuilt and would have no idea how to trace lines, desolder, etc. I was hoping it would be something software wise or  maybe a simple trick to fix it. Long shot in the dark, I know. Thanks anyways.

tompi, 2025-03-04:
Like i said. I think its something simple like a piece of metal stuck in the case or between plates. Open it up and have a look

tompi, 2025-04-20:
Sorry, not really sure what the problem could be, would probably contact the builder...

Alternatively, if the MCU is socketed, try replacing it(its very cheap, like 4$ i think...)

tompi, 2025-04-22:
One thing just occured to me, when this happened, did you use a magnetic cable?
(if so, did you also get messages about excessive power draw from usb device?)

closed 2025-08-22 by tompi