tompi/cheapino #134: M and right key of right thumb cluster not working

LMT1337, 2025-03-17:
Hi,
was wondering if somebody can help me with an issue im facing.
Built my cheapino over the weekend with hot swap sockets and normal diodes.

Was using it for 6-7 hours and now the M Key and the right key of the right thumb cluster are not working. What do you guys suggest - i tried soldering a new diode but that didnt help it seems.

tompi, 2025-03-17:
Did you try reflowing the hotswap socket? Also, Are you sure you swapped the correct diodes ?

I assume you tested replacing the switch?

And does it trigger if you short "before the diode" (you should get 2 keypresses if you do this)

LMT1337, 2025-03-18:
I changed the Diode Right next to the key - tested with a few switches, same result.

Will reflow and Check everything - it also did not work with shorting the hot swap socket

tompi, 2025-03-18:
Use this site to track the problematic lines, and try to short them at different points on the pcb, to pinpoint where the problem is:

https://tompi.github.io/cheapino/doc/troubleshooting/routing.html

LMT1337, 2025-03-18:
I just sat down to troubleshoot it, connected it to my notebook and shorted some hot swap sockets to see where the problem lies and it just worked again, i don't know what happened - if it's acting up again i will reflow the hotswap sockets

Thansk for your immediate help!

closed 2025-03-18 by LMT1337