tompi/cheapino #136: 2025/4/1 - Thank you/build photos

mmHi6pYqJYX79Ukuc4A7BJ, 2025-04-01:
Good morning. (Ha! April fools' 🤡 )
Thank you for providing schematics and guides on how to build your own split keyboard. 🥇 
It was my first (successful) solder project, and it was very fun experiences.👍 
Learning new layouts (Miryoku).
Hoping this helps with the wrist/forearm pinches.

Please see attached for pictures ⌨:

[image]

[image]

[image]

Side notes/feedback 😃: 
I am not sure if it was the 3d casing (the white one) but I bent the legs on 4-5 switches while pushing into the case/hot swap.
It was with the silent peach v2 switches, and same leg out of the two bent - straightened with a pliers but won't be using those switches with the 3d case. 
I think the legs were thinner than lemon but haven't directly compared yet.
Didn't have issue with the tent cases with lemon switches. 
The flat case also had some differences with dial (the rectangular body vs the hole) but thankfully there were some rooms to work with to cut (break) and file it down.  

Reviewing pictures multiple times (along with Imgur reference) helped.
Almost started the project on wrong foot by putting hot swap on front rather than back.
Instruction was there but didn't make sense until I saw different pictures.
I suppose it's the visual side brain and focused only on how to align it properly.

Almost forgot - it may be helpful to others if there is a link in your guide for creating keyboard layout reference. 
I used this website - process was tedious but very helpful.
I understand if you choose not to, being that it is a 3rd party website.

[image]

Fun thought 🤔:
I can see the 5th spare board for macro purposes, similar to my 3 key keyboard (current macro set as left, up, down but it can do more).
What do you think? :)
Once again, thank you very much.

tompi, 2025-04-20:
Hi, sorry about the late reply, but yes, the case is a bit tight, and sometimes a switch will get a bent pin if you dont angle it correctly while pushing it in, not sure its avoidable...

Nice builds by the way, Ill steal the picture and put it up :)

For a layout-editor, you mean just a visual "all layers in a picture" ?
Ill check if the layout-editor supports qmk json files(or zmk...)

tompi, 2025-04-21:
Hey, looked into the keyboard layout, and you can actually import the vial.json file for cheapino to get the
physical layout(the inner array of this: https://github.com/tompi/vial-qmk/blob/cheapinov2/keyboards/cheapino/keymaps/vial/vial.json):

[image]

But how do you get those nice colors and sub-captions per layers etc? is it all manual?
Could you post your json to generate it?

mmHi6pYqJYX79Ukuc4A7BJ, 2025-04-22:
No, I'm happy to hear back from you.

I generally understood that .json, .zmk, .qmk are extension files of sort for custom keyboard while making cheapino but not too familiar with the ins and outs.

Could you tell me how you extracted the vial.json flie to upload on keyboard-layout-editor.com? 😅 
I used Vial portable (Windows) so that may be why I couldn't find the export option.
Tried USB mode as well.

Yes, layout-editor as in looking at periodic table of elements to understand where certain elements are located, what number they are, etc. within one table.
Otherwise, I would have to print out multiple pages of the same keyboard layout, which makes it difficult in many ways..
I noticed some bloggers had it printed out nicely for them to learn a new layout.

I manually mapped the keys on that website with reference from Miryoku suggestion link.
Thankfully (iirc), the color, once chosen, stays on same location when typed in (ex. middle left = purple).
Font and numbers of letters required manual input as well since characters can overlap or become illegible (too small to read).

Here's the .json file I downloaded from that website
[miryoku.json](https://github.com/user-attachments/files/19851802/miryoku.json)
Exported my .vil file from Vial.exe (edit: Github doesn't allow .vil file)

mmHi6pYqJYX79Ukuc4A7BJ, 2025-04-22:
Tested my .json file on the website.
Strangely it removed the keycap style and removed layers, Win, Alt, Ctrl, Shift.
Highlighting and choosing Chicklet on 'Profile/Row' brings the information back.

tompi, 2025-04-22:
If you just copy paste line 11 to 31 from https://github.com/tompi/vial-qmk/blob/cheapinov2/keyboards/cheapino/keymaps/vial/vial.json
into the "raw data" section, you will get layout used by https://vial.rocks for the cheapino.

I can see you put a lot of effort in your keymap, and it looks very nice.
Would be fun to have a qmk configurator json syntax to keyboard editor converter,
but I guess a lot of hard choices would need to be preconfigured.

If you dont mind me asking: why did you not color code layer 1?
And would it not be easier to read if you put it top right instead of top center?

Did you use vial to setup the actual keymap? (do you have the file for this as well?)

mmHi6pYqJYX79Ukuc4A7BJ, 2025-04-22:
Hmm, gotcha. So what you're saying is that the raw data from json file on GitHub can be used to get the initial setup.

Iirc, the website was acting up and top legend (layer 0 and 1) wouldn't show color setting when clicked.
I uploaded the .json file and now it's working.
Also black are the most often used keys for me (letters and top row) so they need to stand out from colorful ones.

Top center seemed appropriate to me so my eyes don't have to travel too far between the empty spaces to see what the symbols are (ex. What symbols are available on J? Oh, it's [{ )
It is top right on layer keys b/c not enough space to accommodate to use in the middle due to font size and numbers of letters (ex. Enter and Backspace).

[image]

General patterns I noticed while referring to the previously laid out maps were:

- Easier to read when they are left to right, top to bottom
- Layer keys on left thumb covers right side keys - vice versa
- Keep special keys in same fingers (RAlt included for me b/c I need access to language switch modifier for Windows)

Very silly thing that occurred while I was making this layout was I couldn't find Add Key button for volume knob and had to copy/paste.
Then I noticed the big button on top right after I finished and had it printed on paper 😓 

I also manually configured on Vial.exe - I'm sure there is a faster way than manually typing KC_( ) but very new to this software so trying to type out the raw data myself was too much of a challenge for me.
I just exported the .vil file while having the keyboard connected. The loaded the same .vil file for my second cheapino.
I guess you just change the extension to .txt so that GitHub accepts it.
[Miryoku (7 layers).txt](https://github.com/user-attachments/files/19856737/Miryoku.7.layers.txt)