tompi/cheapino #141: Adding/Enhancement  idea

bdeboer-x, 2025-04-17:
I built this and got 5 boards like many here have done.  So, built 2.5 keyboards, and was wanting the use the last board as kind of a macro pad.  To do this, it would be very handy to have it so you can either mount a RJ45 socket to the left side, or a rotary encoder.  How hard would this be to do?  Thanks again for making a wonderful product, it is amazing to type on.

tompi, 2025-04-20:
Hey, thats a solid idea. Environment friendly ;)

I checked, and there should be room on top of the reversed rj45 socket:

[image]

Keeping the issue for a possible next iteration.

bdeboer-x, 2026-09-19:
[image]

just added the knob to the RJ45 connector.  EC11 encoder, same as the right hand side one.  GP0 is connected to the middle of the 3 pins and one of the sides of the switch.  The other switch side goes to the top round hole, next hole down goes to A of the encoder, next one down goes to encoder B

[image]

and yes, hot snot all over it to hold it in place, but, it does work.  Using Vial to program, just use the right side board to program what it does.

tompi, 2026-09-22:
Nice work!

tompi, 2026-09-24:
@bdeboer-x If you want to add this to tips/tricks or maybe the build guide, id be happy to merge it