tompi/cheapino #142: Interest in PCB sandwich case?

LorandBiro, 2025-04-23:
Hi,

I've designed a PCB top plate for the Cheapino and wanted to check if you'd be interested in this kind of contribution before I put more work into polishing the design and preparing it for a pull request. Right now it's more of a personal mod, but I could adapt it to fit within the repo's structure and document it properly if that’s something you’d consider merging. Here's some photo of the top plate in action:

[image]

[image]

[image]

I plan to create a bottom plate as well to complete the sandwich case. Would love to hear your thoughts!

tompi, 2025-04-24:
Hi, looks cool!

If you do I pull request for this I will merge!

Add some text and links from this page probably: https://github.com/tompi/cheapino/blob/master/doc/plates_and_cases.md

LorandBiro, 2025-04-24:
Great, glad to hear you're interested. I'm still waiting for another round of PCB production and some screws/nuts/spacers from AliExpress to test before I can finalize everything. I'm going to create the draft PR right now and I'll finish the documentation when I tested everything I wanted.

tompi, 2025-05-22:
Hey, I dont think the bag should dictate anything, its just a random case from aliexpress :)

I think it looks good, the main point of the top plate is to stabilize switches in the hotswap sockets...
Maybe it has an impact on sound as well?

As it is now, I guess the bottom plate is optional? 

If its working, go ahead and make a pull request and ill merge it.

LorandBiro, 2025-05-22:
Well, I actually created the bottom plate so I could use the bag... :) It was a bit problematic to carry it without anything on the bottom, because the hotswap sockets would catch on the netting inside the bag and sometimes bend the pins. Yes, the bottom plate is optional. I’d only recommend it if you plan to carry the keyboard frequently. Otherwise, I think the top plate alone provides a really good feel and look.

The shape of the top plate doesn’t affect the sound, but the method of securing the bottom plate (spacers vs. screws and nuts) definitely does. I can’t say for sure which one is better, but overall, I prefer the screws and nuts version.

I updated the PR. Here you can see the updated documentation: https://github.com/tompi/cheapino/blob/65d332c17232dbdc5feab268c2ba8a07ba7bfa22/doc/plates_and_cases.md
(I can get a better/cleaner shot for the bottom plate over the weekend)

closed 2025-08-22 by tompi