tompi/cheapino #145: MCU Link in build guide not working

t0bybr, 2025-05-27:
Hi, 

I'm in th e process of getting the parts for a Cheapino v2 and I noticed that the link to the MCU in the build guide isn't working for me, I get an 404 error.

This MCU seems that it should fit? 

https://de.aliexpress.com/item/1005007455959490.html?spm=a2g0n.productlist.0.0.c9bc6948OETNxN&browser_id=4ffedc1d84e94c98bcf8b723a8fae3b9&aff_trace_key=3e6095cd8c624d6fbca2719865541491-1748233174218-03034-UneMJZVf&aff_platform=msite&m_page_id=ushpnafgicawrsap1970fe8869e2118be3b833f81a&gclid=&pdp_ext_f=%7B%22order%22%3A%2270%22%2C%22eval%22%3A%221%22%7D&pdp_npi=4%40dis%21EUR%212.69%212.69%21%21%213.00%213.00%21%40211b876717483185865016912e223c%2112000040830462107%21sea%21DE%211682327152%21X&algo_pvid=e062162b-e500-4fdf-b4b9-71c32852d22e

Or should I get the blue version?

Thanks in advance.

tompi, 2025-05-27:
Yes, I think it should work, but it seems to be slightly different, e.g. there is no reset button?

If you search for "rp2040 zero" Im guessing you should get the blue as well?

tompi, 2025-05-27:
Just make sure to buy the one without presoldered headers if you want to socket(like you should :)

t0bybr, 2025-05-27:
Thanks, I think I will it try one. Will socket it anyway :) Will tell you the outcome.

t0bybr, 2025-06-04:
I ended up ordering this one (fits the green switches :D) and it's working.

https://de.aliexpress.com/item/1005007763998604.html

t0bybr, 2025-06-04:
[image]

Here it is with my v1 :D The only problem I have with the MCU is that it only works with USB-C to USB-A cables. No idea if I fried a resistor while soldering the pins?

tompi, 2025-08-22:
Sorry about the late reply. Its apparently a known problem using quality usb-c to usb-c , check out the reddit post linked in this issue: https://github.com/tompi/cheapino/issues/78

commit a6442be8 (2025-08-22): Update MCU links to working AliExpress items

Updated RP2040-Zero links to include both confirmed working item IDs: 1005007763998604 and 1005007650325892.

Fixes #145

🤖 Generated with [Claude Code](https://claude.ai/code)
closed 2025-08-22 by tompi