tompi/cheapino #150: Stuck keys, entire column, qwerty keys RFV

schuay, 2025-08-08:
Hi there,

I've gone through the other bug reports but did not find one with similar symptoms. For reference, these also mention stuck keys:

https://github.com/tompi/cheapino/issues/40
https://github.com/tompi/cheapino/issues/138

After my first build attempt, I started from scratch with decent tools. Afaict, the soldering this time went well - no repairs needed, no excessive heating times, no obvious shorts or other problems I can see.

However, a single column of keys is stuck. These will trigger like crazy, mostly all at once, but sometimes one at a time as well. Sample output (colemakdh layout, thus different characters than the qwerty ones below): "dddptddddddptdptddptdptdptdptdddddddddddptdddddddddddd".

* The fourth column from the left, on the left keyboard side. On qwerty layout: keys RFV.
* Reproducible without switches plugged in, and without the right side connected.
* Does NOT trigger when the MCU is disconnected from the board.
* DOES trigger with a different, second MCU as well.
* Sometimes, upon bootup, symptoms are gone for a minute, but reappear.

Again, on visual inspection I don't see obvious soldering problems. What's interesting is that this key column shares an underlying circuit column with the key column to the right (keys TGB), but those behave normally.

I'd appreciate any debugging help.

schuay, 2025-08-08:
> Hi! Sounds strange. If you modded the firmware, maybe try with the one linked on github page to check that the problem is still there?

Yeah I did that, using your cheapino_vial.uf2. 

> 
> Also, i made a visual debug page to help with these cases, did uou check it out?(should be linked from troubleshooting)

Yes, but I don't know how it would help with the symptoms I see.

It gets stranger still: I plugged in the MCU into another left half which I'd previously assembled. That one now also exhibits stuck keys and single presses emitting multiple chars. But completely different ones..

I wonder if my components are faulty (all sourced from AliExpress); or if I managed to fry both MCUs.

schuay, 2025-08-08:
> This happens on left part also?

Yes, *only* on the left.

> Did you mount in a case? If so, metal part may be lodged in case?
> 
> Try without case, and only with left part.

No case.

> Check that you did not solder the bridges on the left part.

Check.

> Another thing which might be the cause is if some of the diodes are mounted in the wrong direction...

I checked multiple times already.

I'm about to start assembling my final PCB into another left-side try, checking after each step for problems. Although I have another batch of diodes from a different manufacturer incoming in a few weeks. I wonder if I should wait for those, just in case my current batch is bad..

schuay, 2025-08-08:
Well, what can I say - the third attempt at a left hand side seems to be working and I finally have one full working cheapino :)

I'm not sure what the problem was with with attempt number two; I was just as careful there, and I still don't see what could be wrong with it. The symptoms sure were strange. 

I guess the MCU can be ruled out, likewise the switches, the case, and the cables. I'd love to find out what the problem was to avoid future mistakes, so if you happen to find anything in the photos please do let me know.

Thank you very much for the your help, and for your work on this.

closed 2025-08-08 by schuay

tompi, 2025-08-08:
Cant see any obvious things... does the old mcu fail on the new pcb?

schuay, 2025-08-08:
> Cant see any obvious things... does the old mcu fail on the new pcb?

Nope, it's plugged into it right now.

tompi, 2025-08-09:
Cool. Ill add that to the troubleshooting guide!